ArduPilot/MethodicConfigurator #9: Vehicle Component Editor issues

rotorman, 2024-05-09:
Very helpful and great tool!

I found two issues with latest master branch version "Vehicle Component Editor" that now validates some values when saving data and starting configuration.

1) it does not allow same serial port to be used for multiple items. Typically this makes sense, but there are corner cases. In my case using mLRS, I would want to pick the same serial port for RC Receiver and Telemetry.

2) When using an AIO FC, under ESC section, I cannot pick from drop down "FC connection" a value suitable for AIO, only SERIAL, Main out and CAN.

amilcarlucas, 2024-05-11:
Thanks for the report.

1. That will be a hard one to implement. Most users do not want that.
2. do You have a suggestion for a suitable AIO value?

rotorman, 2024-05-11:
Regarding first issue - the tool should then allow some option, like "Other" to be selected IMHO.
And regarding second, how about "internal" or "AIO"?
Thx!

commit 00b5de4b (2024-05-11): BUGFIX: fixes issue #9

fixes #9
closed 2024-05-11 by amilcarlucas