ArduPilot/MethodicConfigurator #10: Installation failed error - is there a better way?

Ryanf55, 2024-05-11:
Hey there.

Looks like `install_linux.sh` has some red text failures, but the script succeeds. Is this expected behavior? 

OS: Ubuntu 22.

```bash
ryan@B650-970:~/Dev/ardu_ws/src/MethodicConfigurator$ ./install_linux.sh 
Updating package lists...
[sudo] password for ryan: 
Hit:1 https://download.docker.com/linux/ubuntu jammy InRelease
Hit:2 https://dl.google.com/linux/chrome/deb stable InRelease                                                                                                                                                                   
Hit:3 http://packages.ros.org/ros2-testing/ubuntu jammy InRelease                                                                                                                                                               
Hit:4 http://us.archive.ubuntu.com/ubuntu jammy InRelease                                                                                                                                                  
Get:5 http://us.archive.ubuntu.com/ubuntu jammy-updates InRelease [119 kB]                                                                                                                      
Hit:6 https://apt.kitware.com/ubuntu jammy InRelease                                                                                                                                            
Get:7 http://security.ubuntu.com/ubuntu jammy-security InRelease [110 kB]                                                                                                           
Hit:8 https://apt.kitware.com/ubuntu jammy-rc InRelease                                                                                                                                       
Hit:9 http://us.archive.ubuntu.com/ubuntu jammy-backports InRelease                                                                                                                           
Hit:10 http://packages.osrfoundation.org/gazebo/ubuntu-stable jammy InRelease                                                                                           
Get:11 https://ppa.launchpadcontent.net/deadsnakes/ppa/ubuntu jammy InRelease [18.1 kB]                                             
Hit:12 http://repo.vulcanexus.org/debian jammy InRelease                                    
Hit:13 https://ppa.launchpadcontent.net/mosquitto-dev/mosquitto-ppa/ubuntu jammy InRelease
Get:14 https://ppa.launchpadcontent.net/deadsnakes/ppa/ubuntu jammy/main amd64 Packages [23.8 kB]
Get:15 https://ppa.launchpadcontent.net/deadsnakes/ppa/ubuntu jammy/main i386 Packages [10.3 kB]
Get:16 https://esm.ubuntu.com/apps/ubuntu jammy-apps-security InRelease [7,553 B]
Get:17 https://esm.ubuntu.com/apps/ubuntu jammy-apps-updates InRelease [7,456 B]
Get:18 https://download.opensuse.org/repositories/Emulators:/Wine:/Debian/xUbuntu_22.04 ./ InRelease [1,559 B]
Get:19 https://esm.ubuntu.com/infra/ubuntu jammy-infra-security InRelease [7,450 B]
Get:20 https://esm.ubuntu.com/infra/ubuntu jammy-infra-updates InRelease [7,449 B]                                                                                                                                              
Fetched 313 kB in 6s (50.5 kB/s)                                                                                                                                                                                                
Reading package lists... Done
W: https://download.opensuse.org/repositories/Emulators:/Wine:/Debian/xUbuntu_22.04/./InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
Installing Python3 PIL.ImageTk...
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following package was automatically installed and is no longer required:
  nvidia-firmware-535-535.154.05
Use 'sudo apt autoremove' to remove it.
The following additional packages will be installed:
  python3-pil
Suggested packages:
  python-pil-doc
The following packages will be upgraded:
  python3-pil python3-pil.imagetk
2 upgraded, 0 newly installed, 0 to remove and 678 not upgraded.
Need to get 429 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 http://us.archive.ubuntu.com/ubuntu jammy-updates/universe amd64 python3-pil.imagetk amd64 9.0.1-1ubuntu0.3 [9,616 B]
Get:2 http://us.archive.ubuntu.com/ubuntu jammy-updates/main amd64 python3-pil amd64 9.0.1-1ubuntu0.3 [419 kB]
Fetched 429 kB in 0s (987 kB/s)      
(Reading database ... 446733 files and directories currently installed.)
Preparing to unpack .../python3-pil.imagetk_9.0.1-1ubuntu0.3_amd64.deb ...
Unpacking python3-pil.imagetk:amd64 (9.0.1-1ubuntu0.3) over (9.0.1-1ubuntu0.2) ...
Preparing to unpack .../python3-pil_9.0.1-1ubuntu0.3_amd64.deb ...
Unpacking python3-pil:amd64 (9.0.1-1ubuntu0.3) over (9.0.1-1ubuntu0.2) ...
Setting up python3-pil:amd64 (9.0.1-1ubuntu0.3) ...
Setting up python3-pil.imagetk:amd64 (9.0.1-1ubuntu0.3) ...
Uninstalling serial and pyserial...
WARNING: Skipping serial as it is not installed.
Found existing installation: pyserial 3.5
Uninstalling pyserial-3.5:
ERROR: Exception:
Traceback (most recent call last):
  File "/usr/lib/python3.10/shutil.py", line 816, in move
    os.rename(src, real_dst)
PermissionError: [Errno 13] Permission denied: '/usr/lib/python3/dist-packages/pyserial-3.5.egg-info' -> '/tmp/pip-uninstall-fn9e4s_1'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ryan/.local/lib/python3.10/site-packages/pip/_internal/cli/base_command.py", line 180, in exc_logging_wrapper
    status = run_func(*args)
  File "/home/ryan/.local/lib/python3.10/site-packages/pip/_internal/commands/uninstall.py", line 105, in run
    uninstall_pathset = req.uninstall(
  File "/home/ryan/.local/lib/python3.10/site-packages/pip/_internal/req/req_install.py", line 727, in uninstall
    uninstalled_pathset.remove(auto_confirm, verbose)
  File "/home/ryan/.local/lib/python3.10/site-packages/pip/_internal/req/req_uninstall.py", line 380, in remove
    moved.stash(path)
  File "/home/ryan/.local/lib/python3.10/site-packages/pip/_internal/req/req_uninstall.py", line 271, in stash
    renames(path, new_path)
  File "/home/ryan/.local/lib/python3.10/site-packages/pip/_internal/utils/misc.py", line 358, in renames
    shutil.move(old, new)
  File "/usr/lib/python3.10/shutil.py", line 834, in move
    rmtree(src)
  File "/usr/lib/python3.10/shutil.py", line 725, in rmtree
    _rmtree_safe_fd(fd, path, onerror)
  File "/usr/lib/python3.10/shutil.py", line 681, in _rmtree_safe_fd
    onerror(os.unlink, fullname, sys.exc_info())
  File "/usr/lib/python3.10/shutil.py", line 679, in _rmtree_safe_fd
    os.unlink(entry.name, dir_fd=topfd)
PermissionError: [Errno 13] Permission denied: 'requires.txt'
```

amilcarlucas, 2024-05-11:
Hey @Ryanf55 ,

Thanks for reporting.
I will take a look at it on Monday or Tuesday.
On Linux it also works without installing, as long as all dependencies are present .

amilcarlucas, 2024-05-13:
to install all dependencies do:

```bash
pip install MethodicConfigurator
```

amilcarlucas, 2024-05-13:
Yes, pip should be fully functional now and the readme is updated.