ArduPilot/MethodicConfigurator #21: No intermediate parameter files found

Jai-GAY, 2024-06-05:
Version 0.6.5

Which step did I do wrongly? I have created a vehicle directory from the template but not in the default (destination) base directory.
for example C:\data\test\
when I used the last option, "[Open Last Used Vehicle Directory](https://github.com/ArduPilot/MethodicConfigurator/blob/master/USERMANUAL.md#option-3-use-the-last-used-vehicle-configuration-directory)" it said could not find the intermediate files. I seem to [experience the same as this user](https://discuss.ardupilot.org/t/new-ardupilot-methodic-configurator-gui/115038/56) and [this user](https://discuss.ardupilot.org/t/new-ardupilot-methodic-configurator-gui/115038/75).

anything to do with "\\" or "/" in the path?

I was trying to plot the [graph after IMU temp calibration](https://discuss.ardupilot.org/t/how-to-methodically-configure-and-tune-any-arducopter/110842#h-23-configure-imu-temperature-calibrationhttpsardupilotorgcopterdocscommon-imutempcalhtml-using-ardupilot-methodic-configuratorhttpsgithubcomardupilotmethodicconfigurator-7).

```
Edition	Windows 10 Pro
Version	22H2
Installed on	‎13/‎03/‎2021
OS build	19045.4412
Experience	Windows Feature Experience Pack 1000.19056.1000.0
```
```
2024-06-05 13:27:40,056 - INFO - Available connection ports are:
2024-06-05 13:27:40,056 - INFO - COM6 - u-blox Virtual COM Port (COM6)
2024-06-05 13:27:40,058 - INFO - COM28 - Cube Orange Mavlink (COM28)
2024-06-05 13:27:40,059 - INFO - COM5 - Silicon Labs CP210x USB to UART Bridge (COM5)
2024-06-05 13:27:40,059 - INFO - COM3 - u-blox GNSS Receiver (COM3)
2024-06-05 13:27:40,059 - INFO - COM29 - Cube Orange SLCAN (COM29)
2024-06-05 13:27:40,059 - INFO - COM4 - Intel(R) Active Management Technology - SOL (COM4)
2024-06-05 13:27:40,060 - INFO - tcp:127.0.0.1:5760 - tcp:127.0.0.1:5760
2024-06-05 13:27:40,060 - INFO - udp:127.0.0.1:14550 - udp:127.0.0.1:14550
2024-06-05 13:27:40,060 - INFO - Will connect to COM28
2024-06-05 13:27:41,150 - INFO - Autopilot type ArduPilot - Plane/Copter/Rover/Sub/Tracker, https://ardupilot.org
2024-06-05 13:27:41,151 - INFO - Vehicle type Hexarotor running ArduCopter firmware
2024-06-05 13:27:41,661 - INFO - Autopilot supports MISSION float message type.
2024-06-05 13:27:41,661 - INFO - Autopilot supports the new param float message type.
2024-06-05 13:27:41,661 - INFO - Autopilot supports MISSION_ITEM_INT scaled integer message type.
2024-06-05 13:27:41,662 - INFO - Autopilot supports COMMAND_INT scaled integer message type.
2024-06-05 13:27:41,662 - INFO - Autopilot supports the File Transfer Protocol v1: https://mavlink.io/en/services/ftp.html.
2024-06-05 13:27:41,662 - INFO - Autopilot supports commanding attitude offboard.
2024-06-05 13:27:41,662 - INFO - Autopilot supports commanding position and velocity targets in local NED frame.
2024-06-05 13:27:41,663 - INFO - Autopilot supports commanding position and velocity targets in global scaled integers.
2024-06-05 13:27:41,663 - INFO - Autopilot supports terrain protocol / data handling.
2024-06-05 13:27:41,663 - INFO - Autopilot supports the MAV_CMD_DO_FLIGHTTERMINATION command (flight termination).
2024-06-05 13:27:41,663 - INFO - Autopilot supports onboard compass calibration.
2024-06-05 13:27:41,663 - INFO - Autopilot supports MAVLink version 2.
2024-06-05 13:27:41,664 - INFO - Autopilot supports mission fence protocol.
2024-06-05 13:27:41,664 - INFO - Autopilot supports mission rally point protocol.
2024-06-05 13:27:41,664 - INFO - Flight Controller Version: 4.5.0 official
2024-06-05 13:27:41,664 - INFO - Flight Controller HW / board version: 9175040
2024-06-05 13:27:41,664 - INFO - Flight Controller first 8 hex bytes of the FC git hash: 53ad2c2a
2024-06-05 13:27:41,664 - INFO - Flight Controller first 8 hex bytes of the ChibiOS git hash: 6a85082c
2024-06-05 13:27:41,666 - INFO - Flight Controller board vendor: Hex
2024-06-05 13:27:41,666 - INFO - Flight Controller board product: CubeOrange
2024-06-05 13:27:41,667 - INFO - Vehicle type not set explicitly, auto-detected ArduCopter.
2024-06-05 13:27:41,676 - INFO - Configuration steps 'ArduCopter_configuration_steps.json' loaded from C:\Program Files (x86)\MethodicConfigurator\_internal\MethodicConfigurator.
2024-06-05 13:27:56,748 - INFO - Configuration steps 'ArduCopter_configuration_steps.json' loaded from C:\Program Files (x86)\MethodicConfigurator\_internal\MethodicConfigurator.
2024-06-05 13:30:32,261 - INFO - Configuration steps 'ArduCopter_configuration_steps.json' loaded from C:\Program Files (x86)\MethodicConfigurator\_internal\MethodicConfigurator.
2024-06-05 13:31:02,805 - INFO - Configuration steps 'ArduCopter_configuration_steps.json' loaded from C:\Program Files (x86)\MethodicConfigurator\_internal\MethodicConfigurator.
```
[image]

amilcarlucas, 2024-06-05:
You need to double click the PX_F550_CubeBlueH7 directory and enter it.
It is not enough to select the directory.

The error message is explicit. It says the
"No intermediate parameter files found in the selected .....\vehicle" It sould say instead:
"No intermediate parameter files found in the selected .....\vehicle\PX_F550_CubeBlueH7"

You did not enter the directory, hence it can not find files inside it.

Jai-GAY, 2024-06-07:
I already did the double-click into the directory, but the result is the same.

How do users change the "Open last used vehicle configuration directory" path highlighted in yellow?
[image]

amilcarlucas, 2024-06-07:
You need to use "Open an existing vehicle configuration directory" and once you do that, it will use it as *Last used vehicle configuration directory* 

You need to click the "..." button on the "Open an existing vehicle configuration directory" section.

Jai-GAY, 2024-06-11:
I [followed this step](https://github.com/ArduPilot/MethodicConfigurator/issues/23#issuecomment-2154628827) and also solved my double-click issue.

closed 2024-06-11 by Jai-GAY