ArduPilot/MethodicConfigurator #23: PermissionError: [Errno 13] Permission denied

Jai-GAY, 2024-06-07:
v 0.6.6
```
Edition	Windows 10 Pro
Version	22H2
Installed on	‎13/‎03/‎2021
OS build	19045.4412
Experience	Windows Feature Experience Pack 1000.19056.1000.0
```
```
2024-06-07 09:16:00,322 - INFO - Available connection ports are:
2024-06-07 09:16:00,322 - INFO - COM6 - u-blox Virtual COM Port (COM6)
2024-06-07 09:16:00,323 - INFO - COM28 - Cube Orange Mavlink (COM28)
2024-06-07 09:16:00,323 - INFO - COM5 - Silicon Labs CP210x USB to UART Bridge (COM5)
2024-06-07 09:16:00,323 - INFO - COM3 - u-blox GNSS Receiver (COM3)
2024-06-07 09:16:00,323 - INFO - COM29 - Cube Orange SLCAN (COM29)
2024-06-07 09:16:00,323 - INFO - COM4 - Intel(R) Active Management Technology - SOL (COM4)
2024-06-07 09:16:00,324 - INFO - tcp:127.0.0.1:5760 - tcp:127.0.0.1:5760
2024-06-07 09:16:00,324 - INFO - udp:127.0.0.1:14550 - udp:127.0.0.1:14550
2024-06-07 09:16:00,324 - INFO - Will connect to COM28
2024-06-07 09:16:00,381 - INFO - Autopilot type ArduPilot - Plane/Copter/Rover/Sub/Tracker, https://ardupilot.org
2024-06-07 09:16:00,381 - INFO - Vehicle type: Hexarotor running ArduCopter firmware
2024-06-07 09:16:00,769 - INFO - Firmware Version: 4.5.0 official
2024-06-07 09:16:00,770 - INFO - Firmware first 8 hex bytes of the FC git hash: 53ad2c2a
2024-06-07 09:16:00,770 - INFO - Firmware first 8 hex bytes of the ChibiOS git hash: 6a85082c
2024-06-07 09:16:00,771 - INFO - Flight Controller HW / board version: 9175040
2024-06-07 09:16:00,771 - INFO - Flight Controller USB vendor ID: Hex (0x2DAE)
2024-06-07 09:16:00,771 - INFO - Flight Controller USB product ID: CubeOrange (0x1016)
2024-06-07 09:16:01,165 - INFO - MAVFTP is supported by the COM28 flight controller, but not yet from this SW
2024-06-07 09:16:06,281 - INFO - Configuration steps 'ArduCopter_configuration_steps.json' loaded from C:\Program Files (x86)\MethodicConfigurator\_internal\MethodicConfigurator.
2024-06-07 09:16:23,596 - INFO - Configuration steps 'ArduCopter_configuration_steps.json' loaded from C:\Program Files (x86)\MethodicConfigurator\_internal\MethodicConfigurator.
Exception in Tkinter callback
Traceback (most recent call last):
  File "tkinter\__init__.py", line 1921, in __call__
  File "frontend_tkinter_directory_selection.py", line 171, in on_select_directory
  File "backend_filesystem.py", line 549, in store_recently_used_vehicle_dir
  File "backend_filesystem.py", line 511, in __set_settings_from_dict
PermissionError: [Errno 13] Permission denied: 'C:\\Users\\XXXXX\\AppData\\Roaming\\.ardupilot_methodic_configurator\\settings.json'
2024-06-07 09:16:25,138 - INFO - Configuration steps 'ArduCopter_configuration_steps.json' loaded from C:\Program Files (x86)\MethodicConfigurator\_internal\MethodicConfigurator.
```

amilcarlucas, 2024-06-07:
Delete the C:\\Users\\XXXXX\\AppData\\Roaming\\.ardupilot_methodic_configurator directory and re-start the tool.

Jai-GAY, 2024-06-11:
> Delete the C:\Users\XXXXX\AppData\Roaming\.ardupilot_methodic_configurator directory

These are the cause of my two problems. FYI. The top is old, bottom is new.

[image]

closed 2024-06-11 by Jai-GAY