ArduPilot/MethodicConfigurator #47: No vehicle_templates folder

alexandru-g, 2024-09-10:
**Describe the bug**
Fresh install of MethodicConfigurator on Linux Mint using miniconda. There is no `miniconda3/lib/python3.10/site-packages/MethodicConfigurator/vehicle_templates` folder and I cannot start using the software (it needs to create a config for my copter and throws an error that there is no templates folder. I've searched for that folder in case it was a python issue but there wasn't one anywhere.

**To Reproduce**
Steps to reproduce the behavior:

1. Start MethodicConfigurator
2. Click on 'Create vehicle configuration directory from template'
4. See error

**Expected behavior**
The `vehicle_templates` folder should be present.

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**

- OS: Linux Mint
- Version: 22

**ArduPilot context**

- ArduPilot version: compiled from master
- Flight controller hardware: JHEMCU GHF745AIO (ported Ardupilot for it)

alexandru-g, 2024-09-10:
For my use I just copied the folder from the repo to that location.

amilcarlucas, 2024-09-11:
Yes, that is a known issue, the source pip package does contain the files but the binary pip package does not. PR's are welcome.
The solution you found is the same I use :-P

amilcarlucas, 2024-10-30:
A workaround is:

```bash
git clone https://github.com/ArduPilot/MethodicConfigurator.git
cd MethodicConfigurator
./install_linux.sh
```

commit 914e510f (2024-11-12): FIX: packaging and accessing locales and vehicle templates on linux

fixes #47
closed 2024-11-12 by amilcarlucas