ArduPilot/MethodicConfigurator #63: Support Flow Sensors

WARGRaph, 2024-11-04:
### Is there an existing issue or pull request for this?

- [X] I have searched the existing issues and pull requests

### Feature description

Configuration and calibration for flow sensors is currently not supported by the methodic configurator, supporting it would be useful for some users

### Desired solution

Add a template file and guide step for flow sensors

### Alternatives considered

None.

### Additional context

https://ardupilot.org/copter/docs/common-optical-flow-sensor-setup.html
https://ardupilot.org/copter/docs/common-optical-flow-sensors-landingpage.html

amilcarlucas, 2024-11-04:
Add two files to your vehicle configuration directory:

file `51_opt_flow_setup.param` :
```
FLOW_TYPE,6  # we have a dronecan flow sensor
FLOW_POS_X,0
FLOW_POS_Y,0
FLOW_POS_Z,0
RC8_OPTION,158 # Use RC8 channel to enable Optflow Calibration
```
file `52_opt_flow_results.param` :
```
FLOW_FXSCALER,0
FLOW_FYSCALER,0
```

amilcarlucas, 2024-11-05:
Download and use version 0.9.9 or newer. It includes an improved version of this feature.

PR #64 (2024-11-04): FEATURE: Add optional optical flow calibration support
closes #63