ArduPilot/MethodicConfigurator #68: Tempcal is confusing

WARGRaph, 2024-11-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Description of the bug

When using the MethodicConfigurator software to run the temperature calibration, the GUI doesn't show the parameters the command line puts out. 
The file written to disk is correct. 

### Steps To Reproduce

- Run Tempcal with MethodicConfigurator 
- Compare values in GUI and on file

### Expected behavior

- Show the freshly calculated values

### Screenshots / Logs

[image]

### Software information

- Operating system: Win10
- arduPilot Methodic Configuration version: 0.9.9

amilcarlucas, 2024-11-12:
https://github.com/ArduPilot/MethodicConfigurator/commit/1c55d17f239501ae1d8fa80f6e405353bea60827

might have fixed it.

WARGRaph, 2024-11-13:
Tried it on windows:

```
Exception in Tkinter callback
Traceback (most recent call last):
  File "tkinter\__init__.py", line 1968, in __call__
  File "frontend_tkinter_parameter_editor.py", line 560, in on_param_file_combobox_change
  File "frontend_tkinter_parameter_editor.py", line 473, in __do_tempcal_imu
  File "tempcal_imu.py", line 471, in IMUfit
  File "tempcal_imu.py", line 532, in generate_calibration_file
AttributeError: module 'os' has no attribute 'fdatasync'
```

occurred after the tempcal was written to file.

amilcarlucas, 2025-01-11:
No feedback for almost two months. Closing it for now. Do re-open it if the problem persists.

closed 2025-01-11 by amilcarlucas