ArduPilot/MethodicConfigurator #174: Please clarify use on already-tuned machines, and other GCS (general feedback and questions)

AndKe, 2025-02-21:
### Is there an existing issue or pull request for this?

- [x] I have searched the existing issues and pull requests

### Feature description

After reading https://ardupilot.github.io/MethodicConfigurator/INSTALL.html and the main page I still have some questions:

1: is it ok to run this process on a previously tuned/flying copter, or does the process depend on starting with AC defaults and may be thrown off by having an existing configuration based on older tuning methods?

2: the documentation is clear on using updated Mission Planner.  I do not involve windows in things that fly, I could run MissionPlanner using wine, but used to be even less stable.   Question: cab QGC or MAVProxy be used instead? - is MP maybe just used to share the MAvLink connection with MethodicConfigurator?  - or is there a deeper cooperation between those two?

Thank you for the app, looking forward to test it.

### Desired solution

-

### Alternatives considered

-

amilcarlucas, 2025-02-21:
Those are support questions and not software issues.

1. yes, it is OK. It's also documented in the USECASES.md file. But yes, if you messed up some very strange parameters in you configuration, the software will not correct them for you.
2. I do not recommend it. MP is only required for step number 12 all other steps do not require MP. Just use it on wine.

closed 2025-02-21 by amilcarlucas

amilcarlucas, 2025-02-21:
I asked you questions to our [ArduPilot AI guru]( https://gurubase.io/g/ardupilot) and these are the answers:

1. https://gurubase.io/g/ardupilot/run-ardupilot-methodic-configurator-on-tuned-copter
2. https://gurubase.io/g/ardupilot/can-qgc-or-mavproxy-replace-mission-planner/binge/3cb5ca5c-5b94-47b1-9307-ea3301a2cd24

AndKe, 2025-02-21:
Perfect, and thanks for the interesting AI  - I will surely try it.
have a great weekend.