ArduPilot/MethodicConfigurator #247: Manufacturer field incorrect

peterbarker, 2025-03-09:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Description of the bug

[image]

Contents of manufacturer field looks very wrong; should be simply "Holybro"

### Steps To Reproduce

Fire it up and proceed through to pictured step.  Also appears on a previous step where the window magically disappeared on me.

### Expected behavior

Don't do that

### Screenshots / Logs

As above

### Software information

- Operating system: Linux
- arduPilot Methodic Configuration version: master

### Additional context

There are *way* too many fields to fill in to make a trivial bug report...

peterbarker, 2025-03-09:
... actually, all of this is garbage - it's not a Matek board:

[image]

amilcarlucas, 2025-03-09:
Thanks for the feedback Peter.

The problem is that the board ID is not an injective function ( not 1-to-1). The issue is visible in the [middleware_fc_ids.py](ardupilot_methodic_configurator/middleware_fc_ids.py) file. Many APJ_BOARD_ID reuse the same number. I am open to suggestions on how to solve this.

amilcarlucas, 2025-03-09:
I plan to transform the "Manufacturer" field to a combobox with the manufacturers from that string.
In other words:

- if the APJ_BOARD_ID can be traced to a single manufacturer, fill that in as read-only and be done with it.
- if the APJ_BOARD_ID is used by multiple manufacturers, present them in a combobox for the user to select the correct one. Once he does, store it in the file.

Would that be OK for you?

amilcarlucas, 2025-06-15:
I get no answer and with the new "simple" mode this no longer gets displayed top the user. So closing for now.

Re-open if it still bothers you in the future.