ArduPilot/MethodicConfigurator #265: Throwing error for duplicate none

Yuvrajscheema, 2025-03-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Description of the bug

Selecting the connection type to the FC as None for two different components gives a duplicate item error, which I believe is a bug. I think it can be fixed by adding ```"None"``` to the if statement at line 737 in the file frontend_tkinter_component_editor.py.

### Steps To Reproduce

Run the configurator and select None as the connection for more than one FC connection and try to continue.

### Expected behavior

You should be able to have to FC connections as None if the components do not exist on the copter.

### Software information

- Operating system: Ubuntu
- arduPilot Methodic Configuration version: 1.2.1

amilcarlucas, 2025-03-11:
Thanks for your Analysis. It looks correct.

amilcarlucas, 2025-03-11:
Please test if the fix works for you:
https://github.com/ArduPilot/MethodicConfigurator/actions/runs/13793422842/artifacts/2731566273

Yuvrajscheema, 2025-03-11:
Could you provide this for a linux system?

amilcarlucas, 2025-03-12:
No, for linux you need to checkout the branch localy.

amilcarlucas, 2025-03-12:
I tested it, it works. So merged.

commit e2170ffd (2025-03-11): fix(component editor): multiple components can have connection "None"

So do not flag it as duplicated
fixes #265
commit ee8801e0 (2025-03-12): fix(component editor): multiple components can have connection "None"

So do not flag it as duplicated
fixes #265
closed 2025-03-12 by amilcarlucas