ArduPilot/MethodicConfigurator #459: Installing .\SetupDeveloperPC.bat on Windows result in Avast warning

flyasky, 2025-05-02:
[image]

    github.com/zricethezav/gitleaks/v8: open C:\Users\***\AppData\Local\Temp\go-build835892825\b001\exe\a.out.exe: Operation did not complete successfully because the file contains a virus or potentially unwanted software.

amilcarlucas, 2025-05-03:
That script is only for *ArduPilot methofic configurator* developers. Normal users should not use the script.

Do you want to help develop the *ArduPilot methofic configurator* software?

It looks like the issue is on the gitleaks tool:

- [gitleaks](https://github.com/gitleaks/gitleaks): a tool for detecting secrets like passwords, API keys, and tokens in git repos.

If you want to help, report this issue to them.

flyasky, 2025-05-10:
I guess, we could close it - they have similar issue in their repo: https://github.com/gitleaks/gitleaks/issues/906

closed 2025-05-10 by flyasky