ArduPilot/MethodicConfigurator #460: Scroll with mouse wheel does not work over labels and text fields in components and in parameters

flyasky, 2025-05-02:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Description of the bug

AMC 1.4.4
On Windows scroll is possible only on empty places without any child component. 

### Steps To Reproduce

No steps - it's just existing UI bug

### Expected behavior

Would be handy to enable scrolling over all components except comboboxes.

### Software information

- Operating system:
- arduPilot Methodic Configuration version:

amilcarlucas, 2025-05-03:
I had to disable mouse scrolling because it was wreaking havoc on comboboxes.
It would be great to re-enable, but keep comboboxes disabled

Siddharthh39, 2025-05-26:
Is this issue still on?

amilcarlucas, 2025-07-03:
See ardupilot_methodic_configurator/frontend_tkinter_component_editor.py:192-219

Trishanshv, 2025-08-29:
@amilcarlucas if this issue is still open
Would it be okay if I take this up?

Trishanshv, 2025-09-02:
@amilcarlucas, 
This is my first contribution so I wanted to share my approach before implementing it.  

I’m thinking of creating a small helper function that forwards mouse wheel events from entries and comboboxes back to their parent frame . 
please correct me if I am wrong .

closed 2025-09-04 by amilcarlucas

amilcarlucas, 2025-10-21:
I see a pattern here. On your computer ist stops scrolling whenever it opens a tooltip popup (yellow help text).

I do not remember seeing this on my computer, and I do have a slow windows computer. I will test it again.

PR #764 (2025-09-04): Scroll with mouse wheel does not work over labels and text fields in components and in parameters #460
This PR fixes an issue where the main window would not scroll when the mouse cursor was hovering over child widgets like text entries (ttk.Entry) and labels. #460 
This implementation adopts the improved approach suggested by @amilcarlucas  in the issue comments.
Closes #460.

Note: The test_user_benefits_from_consistent_ui_styling test appears to be failing on the main branch and is unrelated to these changes.