ArduPilot/MethodicConfigurator #490: Segmentation Fault on Initial Configuration Screen Results In Loss Of All Data Entered

ChasinSpin, 2025-05-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Description of the bug

After 15 minutes of entering data into the initial configuration screen, clicked to another window and then back on the "AP" icon in the OSX Toolbar.  AMC segv'd and all the data entered was lost.  See screenshot.

This is on OSX.

[image]

### Steps To Reproduce

Unable to reproduce despite repeating 3 more times.

### Expected behavior

No SegV

### Software information

- Operating system:
- arduPilot Methodic Configuration version:

amilcarlucas, 2025-05-16:
Without the segfault file there is no way for me to fix this. Closing.
Re-open if you can provide the segfault file, or reproducible steps.

closed 2025-05-16 by amilcarlucas

ChasinSpin, 2025-05-16:
Here's the crash report:

`-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               python3.11 [55335]
Path:                  /Users/USER/*/python
Identifier:            python3.11
Version:               ???
Code Type:             X86-64 (Translated)
Parent Process:        bash [48274]
Responsible:           Terminal [10634]
User ID:               501

Date/Time:             2025-05-15 19:37:18.4423 -0600
OS Version:            macOS 14.5 (23F79)
Report Version:        12
Anonymous UUID:        A1A070C8-F988-ADA2-B8B0-F7820D05B1B6

Time Awake Since Boot: 1800000 seconds

System Integrity Protection: enabled

Notes:
PC register does not match crashing frame (0x0 vs 0x111A4E493)

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000170
Exception Codes:       0x0000000000000001, 0x0000000000000170

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [55335]

VM Region Info: 0x170 is not in any region.  Bytes before following region: 4368731792
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      104659000-104a20000    [ 3868K] r-x/r-x SM=COW  /Users/USER/*/python3.11

Error Formulating Crash Report:
PC register does not match crashing frame (0x0 vs 0x111A4E493)

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libtcl8.6.dylib               	       0x111a4e493 Tcl_FindCommand + 99
1   libtk8.6.dylib                	       0x1116b6ee7 -[TKApplication(TKHLEvents) handleReopenApplicationEvent:withReplyEvent:] + 87
2   Foundation                    	    0x7ff80e90424e -[NSAppleEventManager dispatchRawAppleEvent:withRawReply:handlerRefCon:] + 308
3   Foundation                    	    0x7ff80e904061 _NSAppleEventManagerGenericHandler + 80
4   AE                            	    0x7ff814b477be 0x7ff814b3c000 + 47038
5   AE                            	    0x7ff814b4700c 0x7ff814b3c000 + 45068
6   AE                            	    0x7ff814b40961 aeProcessAppleEvent + 409
7   HIToolbox                     	    0x7ff81891e694 AEProcessAppleEvent + 55
8   AppKit                        	    0x7ff810fedf8a _DPSNextEvent + 1689
9   AppKit                        	    0x7ff811901dc0 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1273
10  libtk8.6.dylib                	       0x1116c3123 TkMacOSXEventsCheckProc + 339
11  libtcl8.6.dylib               	       0x111a5233b Tcl_DoOneEvent + 347
12  _tkinter.cpython-311-darwin.so	       0x10faf0944 _tkinter_tkapp_mainloop_impl + 228
13  python3.11                    	       0x1046dec6c method_vectorcall_FASTCALL + 124
14  python3.11                    	       0x104825ef7 _PyEval_EvalFrameDefault + 238407
15  python3.11                    	       0x1047e923d _PyEval_Vector + 493
16  python3.11                    	       0x1047e8fda PyEval_EvalCode + 250
17  python3.11                    	       0x10489a667 run_mod + 167
18  python3.11                    	       0x10489a475 pyrun_file + 133
19  python3.11                    	       0x104899f53 _PyRun_SimpleFileObject + 275
20  python3.11                    	       0x1048998ff _PyRun_AnyFileObject + 143
21  python3.11                    	       0x1048be73b pymain_run_file_obj + 267
22  python3.11                    	       0x1048be1b5 pymain_run_file + 85
23  python3.11                    	       0x1048bdaa8 Py_RunMain + 1816
[... 994 lines cut ...]
    {
      "rolloutId" : "60186475825c62000ccf5450",
      "factorPackIds" : {

      },
      "deploymentId" : 240000083
    },
    {
      "rolloutId" : "6297d96be2c9387df974efa4",
      "factorPackIds" : {

      },
      "deploymentId" : 240000032
    }
  ],
  "experiments" : [

  ]
},
  "reportNotes" : [
  "PC register does not match crashing frame (0x0 vs 0x111A4E493)"
]
}

`

amilcarlucas, 2025-05-16:
That is very deep in TK library. Sorry, but I do not see how to fix that on my code.