ArduPilot/MethodicConfigurator #492: Ability to set Telemetry to USB and None in Vehicle Component Editor

ChasinSpin, 2025-05-16:
### Is there an existing issue or pull request for this?

- [x] I have searched the existing issues and pull requests

### Feature description

ArduPilot has the ability to run without Telemetry, Telemetry over USB and with telemetry over the RC link on the same serial port as the RC RX.   None of these use cases are supported by Vehicle Component Editor.

### Desired solution

Additional FC Connection options other than SERIAL and CAN.  Suggested additional options are:

NONE
USB

### Alternatives considered

None

amilcarlucas, 2025-05-16:
1.  run without Telemetry is not supported yet.
2. Over USB is strongly not recommended from the ArduPilot development team.
3. The  "telemetry over the RC link on the same serial port as the RC RX" is supported.

I will see about supporting None.

ChasinSpin, 2025-05-17:
Re: 2.  It's common to hook up a drone via USB when configuring on the bench.  There's a use case for No Telemetry and getting the telemetry status via OSD, especially where weight is a concern.  I.E. USB use during configuration and not during flight or pre-flight.

amilcarlucas, 2025-05-18:
When using USB for bench configuration you do not need to change any parameter at all. It works out of the box. That is not even considered to be telemetry.  It is after all local.

amilcarlucas, 2025-06-12:
Fixed by #524

closed 2025-06-12 by amilcarlucas