ArduPilot/MethodicConfigurator #495: vehicle_component.json invalid

ChasinSpin, 2025-05-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Description of the bug

Error in log file, however JSON Lint validates the json.

```
2025-05-17 10:36:15,892 - INFO - MAVFTP is supported by the /dev/cu.usbmodem84201 flight controller
2025-05-17 10:36:22,083 - INFO - Outputted 1110 parameters to complete.param
2025-05-17 10:36:22,083 - INFO - Outputted 1110 parameters to 00_default.param
2025-05-17 10:36:35,642 - ERROR - Invalid vehicle components file '/Users/zawie/Library/Application Support/.ardupilot_methodic_configurator/vehicles/CloudRazor2/vehicle_components.json': Validation error: 1504 is not of type 'string'
2025-05-17 10:36:35,771 - ERROR - Invalid vehicle components file '/Users/zawie/Library/Application Support/.ardupilot_methodic_configurator/vehicles/CloudRazor2/vehicle_components.json': Validation error: 1504 is not of type 'string'
2025-05-17 10:36:36,549 - INFO - Opening URL in browser: https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL_ChibiOS/hwdef/SpeedyBeeF405Mini/README.md
2025-05-17 10:37:43,501 - INFO - Vehicle component data saved successfully.
```

vehicle_components.json:
```
{
    "Format version": 1,
    "Components": {
        "Flight Controller": {
            "Product": {
                "Manufacturer": "ArduPilot,AET,Airvolute,Aocoda-RC-H743Dual,DM,CSKY,Japan Aviation Electronics Industry Ltd.,mRo,Horizon31,Dheeran labs,Qiotek,Sierra Aerospace,SkySakura",
                "Model": "SpeedyBeeF405Mini",
                "URL": "https://www.speedybee.com/speedybee-f405-mini-bls-35a-20x20-stack/",
                "Version": ""
            },
            "Firmware": {
                "Type": "ArduCopter",
                "Version": "4.5.7 official"
            },
            "Specifications": {
                "MCU Series": "STM32F4xx"
            },
            "Notes": ""
        },
        "Frame": {
            "Product": {
                "Manufacturer": "ChasinSpin",
                "Model": "3D Printed - XFrame",
                "URL": "",
                "Version": "2"
            },
            "Specifications": {
                "TOW min Kg": 0.21,
                "TOW max Kg": 0.249
            },
            "Notes": "Tornado Intercept Drone"
        },
        "RC Controller": {
            "Product": {
                "Manufacturer": "Radiomaster",
                "Model": "TX16S",
                "URL": "https://www.radiomasterrc.com/products/tx16s-mark-ii-radio-controller",
                "Version": "MKII"
            },
            "Firmware": {
                "Type": "EdgeTx",
                "Version": "2.11.0-Jolly Mon"
            },
            "Notes": ""
        },
        "RC Transmitter": {
            "Product": {
                "Manufacturer": "Radiomaster",
                "Model": "internal",
                "URL": "https://www.radiomasterrc.com/products/boxer-radio-controller-m2",
                "Version": "MII"
            },
            "Firmware": {
                "Type": "ELRS (internal)",
                "Version": "3.5.5"
            },
            "Notes": "Using the internal RF module"
        },
        "RC Receiver": {
            "Product": {
                "Manufacturer": "Radiomaster",
                "Model": "RP4TD ExpressLRS 2.4Ghz True Diversity Receiver",
                "URL": "https://www.radiomasterrc.com/products/rp4td-expresslrs-2-4ghz-diversity-receiver",
                "Version": "FCC"
            },
            "Firmware": {
                "Type": "ELRS",
                "Version": "3.5.5"
            },
            "FC Connection": {
                "Type": "SERIAL2",
                "Protocol": "CRSF"
            },
            "Notes": "This receiver is on the vehicle and is connected to the flight controller.  57600 baud"
        },
        "Telemetry": {
            "Product": {
                "Manufacturer": "",
                "Model": "",
                "URL": "",
                "Version": ""
            },
            "Firmware": {
                "Type": "",
                "Version": ""
            },
            "FC Connection": {
                "Type": "SERIAL3",
                "Protocol": "MAVLink2"
            },
            "Notes": "NOT USED"
        },
        "Battery Monitor": {
            "Product": {
                "Manufacturer": "SpeedyBee",
                "Model": "SpeedyBee F405 Mini 20x20 Flight Controller",
                "URL": "https://www.speedybee.com/speedybee-f405-mini-bls-35a-20x20-stack/",
                "Version": ""
            },
            "Firmware": {
                "Type": "ArduCopter",
                "Version": "4.5.7"
            },
            "FC Connection": {
                "Type": "Analog",
                "Protocol": "Analog Voltage and Current"
            },
            "Notes": "Voltage is done via the flight controller. Current is done via the ESC."
        },
        "Battery": {
            "Product": {
                "Manufacturer": "DarwinFPV",
                "Model": "3S 1150mAh 60C 11.4V FPV Drone Battery",
                "URL": "https://darwinfpv.com/products/darwinfpv-3s-1150mah-60c-11-4v-fpv-drone-battery",
                "Version": ""
            },
            "Specifications": {
                "Chemistry": "LipoHV",
                "Volt per cell max": 4.35,
                "Volt per cell low": 3.6,
                "Volt per cell crit": 3.5,
                "Number of cells": 3,
                "Capacity mAh": 1150
            },
            "Notes": "Regular High Capacity Flight Battery"
        },
        "ESC": {
            "Product": {
                "Manufacturer": "SpeedyBee",
                "Model": "SpeedyBee F405 Mini BLS 35A 20x20 Stack",
                "URL": "https://www.speedybee.com/speedybee-f405-mini-bls-35a-20x20-stack/",
                "Version": ""
            },
            "Firmware": {
                "Type": "BlueJay",
                "Version": "0.21.0"
            },
            "FC Connection": {
                "Type": "SERIAL5",
                "Protocol": "FETtecOneWire"
            },
            "Notes": "Runs DShot300 on Main out and uses SERIAL5 as extra connection for low rate telemetry.  Set to 24KHz.  115200 baud"
        },
        "Motors": {
            "Product": {
                "Manufacturer": "DarwinFPV",
                "Model": 1504,
                "URL": "https://darwinfpv.com/products/darwinfpv-1504-motor",
                "Version": "3800KV"
            },
            "Specifications": {
                "Poles": 14
            },
            "Notes": ""
        },
        "Propellers": {
            "Product": {
                "Manufacturer": "DarwinFPV",
                "Model": "F4019-2 Folding Propeller",
                "URL": "https://darwinfpv.com/products/darwinfpv-f4019-2-folding-propeller",
                "Version": ""
            },
            "Specifications": {
                "Diameter_inches": 4
            },
            "Notes": "Pitch 1.9inch.  Mounting holes 1.5mm, Center Diameter 5mm. Weight 1.42g"
        },
        "GNSS Receiver": {
            "Product": {
                "Manufacturer": "FlyFishRC",
                "Model": "M10QMC-5883L GPS Module",
                "URL": "https://www.flyfish-rc.com/products/flyfishrc-m10qmc-5883l-gps-module",
                "Version": ""
            },
            "Firmware": {
                "Type": "UBlox",
                "Version": ""
            },
            "FC Connection": {
                "Type": "SERIAL6",
                "Protocol": "uBlox"
            },
            "Notes": "Includes Battery.  57600 Baud"
        }
    },
    "Program version": "1.4.5",
    "Configuration template": "4.5.x-params"
}
```

### Steps To Reproduce

Load the above .json

### Expected behavior

No Error

### Software information

- Operating system: OSX
- arduPilot Methodic Configuration version: 1.4.5

amilcarlucas, 2025-05-18:
That error message is currently just informative. It currently has no negative side effects and can safely be ignored. 

It is caused by the numeric motor model. That string can and will be intrepreted as a number. And that is not schema compliant.

That is a known issue and there is already an unmerged pr addressing it.
The pr uses the schema type to cast the variable type instead of over simplified heuristics. 

Thanks for reporting, I will fix it at some point

amilcarlucas, 2025-06-12:
Fixed by #524

closed 2025-06-12 by amilcarlucas