ArduPilot/MethodicConfigurator #538: Methodic configurator won’t connect via bluetooth

Merga189, 2025-06-09:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Description of the bug

I have DEBIAN 12 running here (and Windows 11 in dual boot and also Windows 11 in Virtualbox under Debian 12) and installed the MethodicConfigurator on all three systems. MP on both Windows 11 systems and QGroundControl on Debian 12.

A remark. The installation on Linux 'Newer Linux distributions with venv' worked like a charm. Only Debian needs **python3** command instead of **python.**

In don't have an issue with connecting by USB cable.

But I am using on my airplanes and also on a copter bluetooth for convenience.

Looks like that on the copter here

[image]

and on the computer side 

[image]

Left a bought master in combination with CP2102 and on the right a DIY master with  a CP2104.

No problem over the years in communication ground station <> model.

Here vbox WIN 11 and groundstation

[image]

Disconnected groundstation and tried to connect to MethodicConfigurator

[image]

Windows 11 (dual boot)

[image]

and DEBIAN 12

[image]

and yes, user is member of those groups

[groups.txt](https://github.com/user-attachments/files/20650728/groups.txt)

Any idea, what could be wrong here?

br

### Steps To Reproduce

See bug description.

### Expected behavior

Would be nice to have a connecting without the hassle with a cable.

### Screenshots / Logs

Put those pics in the bug description. I am a retired mechanical engineer and we had one saying 'A picture tells more than a hundred words....' 

### Software information

- Operating system: DEBIAN 12 and Windows 11
- arduPilot Methodic Configuration version: 1.4.8

### Additional context

....

Merga189, 2025-06-10:
How, I am not a programmer, but if someone could make changes in the code, I'll test.

Was a bit astound, that this way of communication is not working, as bluetooth interfacing using HC modules is in AP documentation.

https://ardupilot.org/copter/docs/common-mission-planner-bluetooth-connectivity.html

amilcarlucas, 2025-06-10:
Yes, that could be it.

To test it, change line 255 to:
                device=self.comport.device, baud=57600, timeout=timeout, retries=retries, progress_callback=progress_callback

Merga189, 2025-06-10:
> or try `--device=/dev/ttyUSB0:57600` as command line parameter

That one is easy, but results in errors

```
.ardupilot_methodic_configurator_venv) kalle@nostromo:~$ ardupilot_methodic_configurator --device=/dev/ttyUSB0:57600
2025-06-10 20:08:15,630 - INFO - Running version: 1.4.8 (git hash: )
2025-06-10 20:08:15,910 - INFO - Already running latest version.
2025-06-10 20:08:15,917 - INFO - Available connection ports are:
2025-06-10 20:08:15,917 - INFO - /dev/ttyS0 - ttyS0
2025-06-10 20:08:15,917 - INFO - /dev/ttyUSB0 - CP2104 USB to UART Bridge Controller - CP2104 USB to UART Bridge Controller
2025-06-10 20:08:15,917 - INFO - tcp:127.0.0.1:5760 - tcp:127.0.0.1:5760
2025-06-10 20:08:15,917 - INFO - udp:127.0.0.1:14550 - udp:127.0.0.1:14550
2025-06-10 20:08:15,917 - INFO - Will connect to /dev/ttyUSB0:57600
Traceback (most recent call last):
  File "/home/kalle/.ardupilot_methodic_configurator_venv/bin/ardupilot_methodic_configurator", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/kalle/.ardupilot_methodic_configurator_venv/lib/python3.11/site-packages/ardupilot_methodic_configurator/__main__.py", line 184, in main
    flight_controller, vehicle_type = connect_to_fc_and_set_vehicle_type(args)
                                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kalle/.ardupilot_methodic_configurator_venv/lib/python3.11/site-packages/ardupilot_methodic_configurator/__main__.py", line 84, in connect_to_fc_and_set_vehicle_type
    error_str = flight_controller.connect(args.device, log_errors=False)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kalle/.ardupilot_methodic_configurator_venv/lib/python3.11/site-packages/ardupilot_methodic_configurator/backend_flightcontroller.py", line 170, in connect
    return self.__create_connection_with_retry(progress_callback=progress_callback, log_errors=log_errors)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kalle/.ardupilot_methodic_configurator_venv/lib/python3.11/site-packages/ardupilot_methodic_configurator/backend_flightcontroller.py", line 254, in __create_connection_with_retry
    self.master = mavutil.mavlink_connection(
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kalle/.ardupilot_methodic_configurator_venv/lib/python3.11/site-packages/pymavlink/mavutil.py", line 1938, in mavlink_connection
    return mavudp(device, source_system=source_system, source_component=source_component, input=input, use_native=use_native)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kalle/.ardupilot_methodic_configurator_venv/lib/python3.11/site-packages/pymavlink/mavutil.py", line 1062, in __init__
    self.port.bind((a[0], int(a[1])))
socket.gaierror: [Errno -2] Name or service not known
(.ardupilot_methodic_configurator_venv) kalle@nostromo:~$ 

```

Your first proposal is a bit difficult for me now to understand, what I have to do?

Downloaded the zip file from https://github.com/ArduPilot/MethodicConfigurator.git, unpacked it here. 

No problem modifying that line 

`device=self.comport.device, baud=57600, timeout=timeout, retries=retries, progress_callback=progress_callback`

BUT what do I have to do next? Build the package? My installation comands ....

```
kalle@nostromo:~$ python3 -m venv .ardupilot_methodic_configurator_venv

kalle@nostromo:~$ source .ardupilot_methodic_configurator_venv/bin/activate

(.ardupilot_methodic_configurator_venv) kalle@nostromo:~$ python -m pip install --upgrade pip

(.ardupilot_methodic_configurator_venv) kalle@nostromo:~$ pip install ardupilot_methodic_configurator
```

That `pip install ardupilot_methodic_configurator` is NOT going to install, what I have with that modified line here.

Merga189, 2025-06-12:
Thank you for your patience. BUT as I stated earlier, I am not a python expert, just a 'normal' user.

**This works here**
```

~$ python3 -m venv .ardupilot_methodic_configurator_venv

~$ source .ardupilot_methodic_configurator_venv/bin/activate

(.ardupilot_methodic_configurator_venv) ~$ python -m pip install --upgrade pip

(.ardupilot_methodic_configurator_venv) ~$ pip install ardupilot_methodic_configurator
```
**Your proposal results in an ERROR in step 'pip install uv'**

```
~$ git clone https://github.com/ArduPilot/MethodicConfigurator.git
Cloning into 'MethodicConfigurator'...
remote: Enumerating objects: 12556, done.
remote: Counting objects: 100% (222/222), done.
remote: Compressing objects: 100% (152/152), done.
remote: Total 12556 (delta 121), reused 91 (delta 65), pack-reused 12334 (from 4)
Receiving objects: 100% (12556/12556), 44.63 MiB | 6.83 MiB/s, done.
Resolving deltas: 100% (8794/8794), done.

~$ cd MethodicConfigurator

~/MethodicConfigurator$ pip install uv
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
```

As far as I understand, that ERROR tells me to do those proposed steps in the isolated 'venv' environment ....

amilcarlucas, 2025-06-12:
Try instead:

```bash
git clone https://github.com/ArduPilot/MethodicConfigurator.git
cd MethodicConfigurator
python3 -m venv .ardupilot_methodic_configurator_venv
source .ardupilot_methodic_configurator_venv/bin/activate
python3 -m pip install -e .[dev]
Now edit the ardupilot_methodic_configurator/backend_flightcontroller.py file
python3 ardupilot_methodic_configurator/__main__.py
```

amilcarlucas, 2025-06-12:
No trouble at all. Thanks for testing.
I'll add configurable baudrate at some point.

amilcarlucas, 2025-06-16:
Try:

```bash
git clone https://github.com/ArduPilot/MethodicConfigurator.git
cd MethodicConfigurator
git checkout baudrate
python3 -m venv .ardupilot_methodic_configurator_venv
source .ardupilot_methodic_configurator_venv/bin/activate
python3 -m pip install -e .[dev]
python3 ardupilot_methodic_configurator/__main__.py --baudrate=57600
```

amilcarlucas, 2025-06-17:
Thanks for reporting and testing. The feature is now merged in master.