ArduPilot/MethodicConfigurator #544: Template selection box does not render correctly on HiDPI displays with scaling (Linux - Debian)

gwvr, 2025-06-09:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Description of the bug

My monitor is 4K, with 3840x2160 resolution. I use 200% scaling in Cinnamon (so Linux Mint users with HiDPI monitors will likely also be affected). GUI doesn't render fonts in some selector windows correctly.

### Steps To Reproduce

1. Buy 4K monitor
2. Set HiDPI/scaling to suit the tiny pixels and make everything readable and pretty
3. Run `ardupilot_methodic_configurator`
4. Click "Skip FC connection.."
5. Click selector under "Source Template directory:"
6. Observe incorrectly spaced text and obscured text

If I set display scaling back to 100% (which makes everything look tiny), then that selector renders correctly.

### Expected behavior

Ideally `ardupilot_methodic_configurator` would play nice with HiDPI/Scaled displays. I realise this may not be easy with the chosen GUI software.

### Screenshots / Logs

Fonts in the template selection screen look like this:

[image]

### Software information

- Operating system: Debian GNU/Linux 12 (bookworm) x86_64 with Cinnamon desktop environment
- arduPilot Methodic Configuration version: 1.4.8

Merga189, 2025-06-10:
Not a problem here

EIZO 27" with

```
~$ inxi -Fz
System:
  Kernel: 6.1.0-37-amd64 arch: x86_64 bits: 64 Desktop: Xfce v: 4.18.1
    Distro: Debian GNU/Linux 12 (bookworm)
Machine:
  Type: Desktop Mobo: ASUSTeK model: PRIME X370-A v: Rev X.0x
    serial: <superuser required> UEFI: American Megatrends v: 6232
    date: 09/29/2024
CPU:
  Info: 8-core model: AMD Ryzen 7 5700G with Radeon Graphics bits: 64
    type: MT MCP cache: L2: 4 MiB
  Speed (MHz): avg: 2446 min/max: 1400/3800 cores: 1: 1400 2: 2399 3: 3223
    4: 1400 5: 3250 6: 1400 7: 3284 8: 3568 9: 1400 10: 1400 11: 3027 12: 1400
    13: 3800 14: 1400 15: 3800 16: 2999
Graphics:
  Device-1: NVIDIA GP108 [GeForce GT 1030] driver: nvidia v: 535.247.01
  Display: x11 server: X.Org v: 1.21.1.7 driver: X: loaded: nvidia
    unloaded: fbdev,modesetting,nouveau,vesa gpu: nvidia
    resolution: 1920x1080~60Hz
  API: OpenGL v: 4.6.0 NVIDIA 535.247.01 renderer: NVIDIA GeForce GT
    1030/PCIe/SSE2
```

[image]

Merga189, 2025-06-10:
Therefore I am with my applications on such a small screen. And with Xfce, because it keeps things simple. Started with Kubuntu, then PCLos, changed to Debian starting with Wheezy, Jessie and Stretch and so on. Had LM one year on a TP some years ago, but LM is too much beating around the bush (for me).

Might find help at https://www.python.org/community/forums/

Found great help there when struggling building AP under Debian from the sources last year .....

amilcarlucas, 2025-06-12:
See file ardupilot_methodic_configurator/frontend_tkinter_template_overview.py line 89 to 134

closed 2025-06-22 by amilcarlucas

gwvr, 2025-07-06:
I suspect Windows and Cinnamon on Linux handle scaling in different ways.

[image]

Linux thinks the monitor is 96dpi. "Monitor Scale" display setting in Cinnamon is set to 200%. Default fonts are 9-11 pt, and "Text Scaling Factor" is 1.0.

```
> xdpyinfo | grep -B 2 resolution
screen #0:
  dimensions:    3840x2160 pixels (1016x571 millimeters)
  resolution:    96x96 dots per inch
```

Actual monitor screen size is approximately 62x34cm. Obviously 62cm != 42 inches | 1016 mm. Actual DPI is roughly 160.

commit 77479b87 (2025-06-23): feat(HiDPI support): Support Hi DPI monitors

resolves #544
Scale fonts and windows according to monitor DPI settings
BEHAVIOR-DRIVEN TESTS - Focus on User Stories and Business Value
Added pytest instructions file for AI agents