ArduPilot/MethodicConfigurator #806: Feedback: Slightly ambiguous information , and  human use vs ideal user.

AndKe, 2025-09-16:
No, the UAV is not "correctly configured" (as per your standards) - which is what I try to achive.

[image]

same wording is used here: https://ardupilot.github.io/MethodicConfigurator/USECASES.html#create-a-vehicle-configuration-based-on-a-correctly-configured-vehicle

Reading this lets me interpret it in two ways:
1: The UAV is expected to be *correctly* configured, no further tuning/changing of filters etc will be done, or the current configuration will be kept as defaults/master/gospel selecting it treats the current configuration as a really good choice. 

2: The UAV is expected to have some basic configuration "correctly" connected motors/configured frame, and it does fly - but I will still get to go thru all the configuration steps, and correct/fix anything.

I chose the interpretation #2 , but that was because I've seen the process before, otherwise, #1 "correctly configurated" would be my preferred choice - as I expect to have some errors in my old config before using this tool.

Scenario: This is a >5yr  old UAV, MTOW 50kg , it's ArduCopter has been upgraded over time. (also replaced the FC to CubeOrange+ , preserving most configuration.)   It flies fine, I do not try to "solve" any particular  issue.  It's been tuned manually, AutoTuned, things have been changed here and there over time.  Pos controller PID and s

So I opened it, and worked thru some steps at the office, skipped steps 17(notch filter) and forward to 23- quick tune.
Curiosity about what can be done at the office vs outside, is normal.
 
[image]

At this point I wonder: 
Q: How can I go back to 18-notch filter setup/test (18 is really done, how to feed results into 19
Also: what will the options really do now? - I have already prepared for/jumped thru 17...23 , if I select CLOSE,- am I still configured for notch filter experiment?
YES; I am responding to "have performed experiment" ... does a working autotune/flying copter qualify for this, or will I mess up future flow as the quick tune script have never been used?
NO; The one "have not performed experiment, but know what you are doing" -  I do not know at this point is continuing with the current config is somehow "bad" for the next steps.  I do not know what I am doing in terms of "will methodic_configurator base next/future changes on this?

I see no option to "go back to step xx and repeat it".   Is the idea that if I, like many users, get this far and skip some steps, one should continue till "the end" (everyday use) - and then will be able to repeat steps?

I know that I have not been following the flow strictly as intended,  and I know that many more users will do this as well, "getting to know the tool" before driving out of the city to a suitable spot, on a day with suitable weather.   It would be very nice to have some information about the consequences of "skipping steps" like this, and the effect of those, like is notch-filter related params not unseat as I went past that step? -and if so, how to get back?

amilcarlucas, 2025-09-16:
Both #1 and #2 are correct. Some parameters will get changed so that the experiments can be performed, and if the results of the experiments match you current tunning (usually they never do), no changes will be done in the result steps.

> So I opened it, and worked thru some steps at the office, skipped steps 17(notch filter) and forward to 23- quick tune.
Curiosity about what can be done at the office vs outside, is normal.

You choose the (default) "simple" GUI complexity. Using this mode you are not supposed to skip anything, and you can not jump back.
Most beginner users mess thinks up by doing things out-of-order and/or skipping steps. Hence the simple (safe) mode does not allow you to mess up.

> Q: How can I go back to 18-notch filter setup/test
Select the "normal" GUI complexity on top of the component editor window. Then you can change the step in the combobox on top of the parameter editor window.

> how to feed results into 19
Enter the values manually as usual.

> I have already prepared for/jumped thru 17...23 , if I select CLOSE,- am I still configured for notch filter experiment?
Yes.

 > YES; I am responding to "have performed experiment" ... does a working autotune/flying copter qualify for this?
Yes.

>  or will I mess up future flow as the quick tune script have never been used?
No.

> NO; The one "have not performed experiment, but know what you are doing" - I do not know at this point is continuing with the current config is somehow "bad" for the next steps. I do not know what I am doing in terms of "will methodic_configurator base next/future changes on this?
No it will not. It makes no changes whatsoever.

> I see no option to "go back to step xx and repeat it". Is the idea that if I, like many users, get this far and skip some steps, one should continue till "the end" (everyday use) - and then will be able to repeat steps?
Almost. The idea is to continue to the end and finish it. Once that is done restart the software, and select the "normal" GUI complexity. Then you can do whatever you want, jump, skip, go back. But there are no guaranties. The idea is to do this stuff in the order that is presented, if you do not, you are on your own.

AndKe, 2025-09-16:
Thank you for the explanation. As you see, new users are most likely to start going too deep out of order - out of curiosity and/or wanting to check out the process before going out to the field. And no, I have no quick fix for the paradox that the very same new users could be "protected" from the "normal GUI" .

Hopefully you get some ideas out of this report, feel free to close the issue unless you want it open for now.